rtv/Stage #48: LTDL_INCLUDE_DIR Cmake error during configure

rashadkm, 2015-02-15:
-- Stage version 4.1.1
-- Operating system is Linux
-- Linux x86_64 Detected
-- Installing Libraries to /home/rashad/opt/lib64
-- Installing Plugins to /home/rashad/opt/lib64/Stage-4.1
-- Build type RELEASE
-- Checking for libtool
-- Checking for required libraries...
-- Checking for optional libraries...
-- checking for one of the modules 'playercore>=2.1.0'
--   * Player not detected. If Player is installed but not detected, check your PKG_CONFIG_PATH.
-- Installation path CMAKE_INSTALL_PREFIX=/home/rashad/opt
-- Configuring libstage
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LTDL_INCLUDE_DIR
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/libstage
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/libstage
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/examples/ctrl
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/worlds/benchmark
   used as include directory in directory /home/rashad/sources/playerstage/github_Stage/worlds/benchmark
LTDL_LIB
    linked by target "stage" in directory /home/rashad/sources/playerstage/github_Stage/libstage

-- Configuring incomplete, errors occurred!
See also "/home/rashad/build/stage/CMakeFiles/CMakeOutput.log".

rtv, 2015-02-15:
Which platform are you installing on?

—
http://autonomylab.org

rashadkm, 2015-02-15:
Operating system is Linux
-- Linux x86_64 Detected

Fedora 20 64 bits

git log gives..
commit f9c59ca77f203049979dcf32223e5d2b26c03bf2
Author: Richard Vaughan <vaughan@sfu.ca>
Date:   Fri Jul 5 15:49:38 2013 -0700

    fixed Player blobfinder interface, per issue #21

rashadkm, 2015-02-15:
yum install -y libtool-ltdl-devel 
fixes the issue. But  why I need -devel package for libtool  for Stage. I already have libtool installed and most of my development seems fine?

rtv, 2015-02-15:
Stage loads dynamic libraries and uses the interface packaged in
libtool-devel. Some systems break up libtool this way, some don't.

rashadkm, 2015-02-15:
So this issue can be closed.

rtv, 2015-02-15:
If that dependency is listed in the docs then yes. If not, I'll leave it open until the docs are fixed.

closed 2017-04-26 by rtv