rtv/Stage #50: The SimulatingOneRobot tutorial is several years out of date

trainman419, 2015-03-29:
The ROS/stage tutorial for simulating a single robot is very, very out of date: http://wiki.ros.org/stage/Tutorials/SimulatingOneRobot

This has generated a significant number of confused users on ROS answers over the past four years.

 * http://answers.ros.org/question/206158/keyboard-teleoperation-for-the-simulatingonerobot-tutorial-on-ros-hydro/
 * http://answers.ros.org/question/9207/teleop_base-deprecated-need-generic-teleop-package/
 * http://answers.ros.org/question/54078/stage-simulatingonerobot-tutorial-is-there-still-a-teleop_base-package/

rtv, 2017-04-05:
That's useful to know, but out of date docs on the ROS site are not really an issue with the Stage repo. I'll leave this open so it's easy to find this info.

trainman419, 2017-05-23:
The issue tracker link on the [ROS stage page](http://wiki.ros.org/stage) points here, and I lack the experience to update these tutorials myself.

These broken tutorials seem to be generating more issues than they're solving. I think they should be updated, marked as deprecated, or completely deleted.