rtv/Stage #56: Simulation data is not shown

marjinal1st, 2015-11-03:
Hello,

I'm using Stage with ROS Jade on Arch Linux (up-to-date). Version is:

```text
$ stage
Stage 4.1.1 

[Stage: done]
```

When I run any Stage simulation, for example **simple.world** example, no simulation data is shown in the screen (**View -> Data** is selected). Also there's no time information in the left-bottom and none of axes is shown:

[image]

**glxinfo** output: https://gist.github.com/marjinal1st/64f1a055810969eb728d

Any ideas?

rtv, 2017-01-10:
I see various visualization errors that vary by platform. I think there are one or more subtle bugs in the GL code that only show up with some drivers/hardware. Fixing this is on the todo-list. Pull requests welcome.

CodeFinder2, 2017-02-06:
I've observed the same issues.  In my experiences, this is related to an "incompatible" mix of stage.hh  (library headers) and the run-time library libstage.so.  Make sure you compile and dynamic link against the same version, e. g. (Bash),

export LD_LIBRARY_PATH=<PATH_TO_YOUR_CORRECT_STAGE_INSTALL_DIR>:$LD_LIBRARY_PATH
run stage... 

I've used stage_ros with the latest Stage (Github) version and forgot to also recompile stage_ros.

rtv, 2017-02-07:
This is an already-open issue #56 rtv/Stage on GitHub. Fixes or guesses
towards fixes are welcome. I'll look into this wrong-library load idea.

Richard/

rtv, 2017-04-26:
Fixed in master.

Still some issue with the text box in the bottom left, but no other known problems.

closed 2017-04-26 by rtv