rtv/Stage #59: ctrl "wander" segfaults after warning  "Model r0.ranger:1 not found"

sloretz, 2015-12-21:
Stage 4.1.1 installed via the ros-jade-stage package segfaults when trying to use the wander controller

    /opt/ros/jade/bin$ ./stage /tmp/test.world 
    Stage 4.1.1 
     [Loading /tmp/test.world]warn: lookup of model name r0.ranger:1: no matching name (/tmp/buildd/ros-jade-stage-4.1.1-6trusty-20151110-0906/libstage/world.cc GetModel)
    warn: Model r0.ranger:1 not found (/tmp/buildd/ros-jade-stage-4.1.1-6trusty-20151110-0906/libstage/model.cc GetChild)
    Segmentation fault (core dumped)

test.world is made from copy and pasted portions of the example worlds

    define p2dxsonar sensor
    (
      size [0.01 0.05 0.01 ] 
      range [0 5.0]
      fov 15
      samples 1
    )
    
    define amigo_sonar ranger
    (
      p2dxsonar ( pose [ 0.073 0.105 0 90 ] )
      p2dxsonar ( pose [ 0.130 0.078 0 41 ] )
      p2dxsonar ( pose [ 0.154 0.030 0 15 ] )
      p2dxsonar ( pose [ 0.154 -0.030 0 -15 ] )
      p2dxsonar ( pose [ 0.130 -0.078 0 -41 ]	)
      p2dxsonar ( pose [ 0.073 -0.105 0 -90 ] )
      p2dxsonar ( pose [ -0.146 -0.060 0 -145 ] )
      p2dxsonar ( pose [ -0.146 0.060 0 145 ] )
    )
    
    define amigobot position
    (
      size [0.330 0.280 0.25]
      origin [0 0 0 0] # what should this value be? send email to vaughan@sfu.ca.
      amigo_sonar( pose [0 0 -0.02 0 ] )
    )
    
    amigobot
    (
      name "r0"
      pose [ -6.946 -6.947 0 45.000 ]
      ctrl "wander"
      localization "gps"
      localization_origin [ 0 0 0 0 ]
    )

The gdb backtrace looks like the following

    Program received signal SIGSEGV, Segmentation fault.
    0x00007ffff7b954f0 in Stg::Model::AddCallback(Stg::Model::callback_type_t, int (*)(Stg::Model*, void*), void*) ()
       from /opt/ros/jade/lib/libstage.so.4.1.1
    (gdb) bt
    #0  0x00007ffff7b954f0 in Stg::Model::AddCallback(Stg::Model::callback_type_t, int (*)(Stg::Model*, void*), void*) ()
       from /opt/ros/jade/lib/libstage.so.4.1.1
    #1  0x00007fffef555122 in Init () from /opt/ros/jade/lib/Stage-4.1/wander.so
    #2  0x00007ffff7b9571a in Stg::Model::CallCallbacks(Stg::Model::callback_type_t) () from /opt/ros/jade/lib/libstage.so.4.1.1
    #3  0x00007ffff7bac9b5 in Stg::World::Load(std::string const&) () from /opt/ros/jade/lib/libstage.so.4.1.1
    #4  0x00007ffff7bc409a in Stg::WorldGui::Load(std::string const&) () from /opt/ros/jade/lib/libstage.so.4.1.1
    #5  0x0000000000401181 in main ()

sloretz, 2015-12-21:
A workaround is to add a second ranger sensor to the robot. The crash happens because the wander controller tries to subscribe to two rangers.

    define amigobot position
    (
      size [0.330 0.280 0.25]
      origin [0 0 0 0] # what should this value be? send email to vaughan@sfu.ca.
      amigo_sonar ( pose [0 0 -0.02 0 ] )
      amigo_sonar( pose [0 0 -0.02 1 ] )    # Added this line
    )

I think it would be useful for the wander controller to write to std_err about needing two rangers when it's asked to control a robot without them.

rtv, 2017-04-26:
Fixed in master.

closed 2017-04-26 by rtv