rtv/Stage #60: ModelCamera ignores x and y components of its pose

sloretz, 2016-04-25:
In Stage-4.1.1 the camera model does not take into account the x and y parts of it's pose. Camera images appear to come from the perspective of the middle of the ModelPosition the camera is on. They should instead be coming from the center of the ModelCamera itself.

The camera below should not be able to see the robot body, but it does.
[image]

Here's a stage world that reproduces the issue
```
interval_sim 50
quit_time 0
resolution 0.02
show_clock 0
show_clock_interval 100
threads 2

window
(
    size [ 600.0 700.0 ]
    center [ 0.0 0.0 ]
    rotate [ 0.0 0.0 ]
    scale 60
)

define o3d3o3 camera
(
    size [ 0.09 0.06 0.06 ]
    color "gray20"
    range [ 0.2 8.0 ]
    resolution [ 176 132 ]
    fov [ 60 40 ]
    pantilt [ 0 65 ]
    alwayson 1
)

define robot_body position
(
    #position properties
    drive "omni"
    localization "gps"
    localization_origin [0 0 0 0]
    watchdog_timeout 0.5

    # Model properties
    size [0.5 0.5 0.5]
    origin [0 0 0 0]
    velocity [0 0 0 0]
    gui_nose 1
    gui_grid 1
    gui_boundary 1
    obstacle_return 1
    ranger_return 1
    color "red"

    o3d3o3
    (
        pose [0.5 0 1.0 0]
    )
)

robot_body
(
    pose [0 0 0 -90]
)

```

AravindaDP, 2017-02-16:
I think following line is where the problem occurs.
https://github.com/rtv/Stage/blob/master/libstage/model_camera.cc#L204
If we can agree to use model pose instead parent pose I think it could be resolved.

AravindaDP, 2017-05-25:
@rtv Any insight on this?

doisyg, 2019-12-26:
I observe the same issue

doisyg, 2019-12-26:
> If we can agree to use model pose instead parent pose I think it could be resolved.

That's indeed the solution, see https://github.com/rtv/Stage/pull/97