rtv/Stage #64: new c++11 string literal breaks macrodefenitions in stage.hh

majorx234, 2016-08-04:
get error when compiling:
<code>
In file included from src/stageros.cpp:38:0:
/opt/ros/indigo/lib/cmake/Stage/../../../include/Stage-4.1/stage.hh: In member function ‘virtual void Stg::Ancestor::SetToken(const string&)’:
/opt/ros/indigo/lib/cmake/Stage/../../../include/Stage-4.1/stage.hh:625:62: error: unable to find string literal operator ‘operator""m’ with ‘const char [24]’, ‘long unsigned int’ arguments
 #define PRINT_ERR(m) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", __FILE__, __FUNCTION__)
                                                              ^
/opt/ros/indigo/lib/cmake/Stage/../../../include/Stage-4.1/stage.hh:725:2: note: in expansion of macro ‘PRINT_ERR’
  PRINT_ERR( "Ancestor::SetToken() called with zero length string. Ignored." );
  ^~~~~~~~~
/opt/ros/indigo/lib/cmake/Stage/../../../include/Stage-4.1/stage.hh: In member function ‘virtual void Stg::Model::SetToken(const string&)’:
/opt/ros/indigo/lib/cmake/Stage/../../../include/Stage-4.1/stage.hh:625:62: error: unable to find string literal operator ‘operator""m’ with ‘const char [24]’, ‘long unsigned int’ arguments
 #define PRINT_ERR(m) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", __FILE__, __FUNCTION__)
                                                              ^
/opt/ros/indigo/lib/cmake/Stage/../../../include/Stage-4.1/stage.hh:2032:2: note: in expansion of macro ‘PRINT_ERR’
  PRINT_ERR( "Model::SetToken() called with zero length string. Ignored." );
  ^~~~~~~~~
make[2]: *** [CMakeFiles/stageros.dir/build.make:63: CMakeFiles/stageros.dir/src/stageros.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:805: CMakeFiles/stageros.dir/all] Error 2
make: *** [Makefile:128: all] Error 2
</code>

the problem is that in c++11 new string literals could be defined (e.g. operator ""m) thus the macrodefinitions won't work anymore

solved it with a patch of libstage/stage.hh:
<code>
@@ -647,46 +647,46 @@
   { cont.erase( std::remove( cont.begin(), cont.end(), thing ), cont.end() ); }
   
   // Error macros - output goes to stderr
-#define PRINT_ERR(m) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", __FILE__, __FUNCTION__)
-#define PRINT_ERR1(m,a) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
-#define PRINT_ERR2(m,a,b) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
-#define PRINT_ERR3(m,a,b,c) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
-#define PRINT_ERR4(m,a,b,c,d) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", a, b, c, d, __FILE__, __FUNCTION__)
-#define PRINT_ERR5(m,a,b,c,d,e) fprintf( stderr, "\033[41merr\033[0m: "m" (%s %s)\n", a, b, c, d, e, __FILE__, __FUNCTION__)
+#define PRINT_ERR(m) fprintf( stderr, "\033[41merr\033[0m: "#m" (%s %s)\n", __FILE__, __FUNCTION__)
+#define PRINT_ERR1(m,a) fprintf( stderr, "\033[41merr\033[0m: "#m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
+#define PRINT_ERR2(m,a,b) fprintf( stderr, "\033[41merr\033[0m: "#m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
+#define PRINT_ERR3(m,a,b,c) fprintf( stderr, "\033[41merr\033[0m: "#m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
+#define PRINT_ERR4(m,a,b,c,d) fprintf( stderr, "\033[41merr\033[0m: "#m" (%s %s)\n", a, b, c, d, __FILE__, __FUNCTION__)
+#define PRINT_ERR5(m,a,b,c,d,e) fprintf( stderr, "\033[41merr\033[0m: "#m" (%s %s)\n", a, b, c, d, e, __FILE__, __FUNCTION__)
 
   // Warning macros
-#define PRINT_WARN(m) printf( "\033[44mwarn\033[0m: "m" (%s %s)\n", __FILE__, __FUNCTION__)
-#define PRINT_WARN1(m,a) printf( "\033[44mwarn\033[0m: "m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
-#define PRINT_WARN2(m,a,b) printf( "\033[44mwarn\033[0m: "m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
-#define PRINT_WARN3(m,a,b,c) printf( "\033[44mwarn\033[0m: "m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
-#define PRINT_WARN4(m,a,b,c,d) printf( "\033[44mwarn\033[0m: "m" (%s %s)\n", a, b, c, d, __FILE__, __FUNCTION__)
-#define PRINT_WARN5(m,a,b,c,d,e) printf( "\033[44mwarn\033[0m: "m" (%s %s)\n", a, b, c, d, e, __FILE__, __FUNCTION__)
+#define PRINT_WARN(m) printf( "\033[44mwarn\033[0m: "#m" (%s %s)\n", __FILE__, __FUNCTION__)
+#define PRINT_WARN1(m,a) printf( "\033[44mwarn\033[0m: "#m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
+#define PRINT_WARN2(m,a,b) printf( "\033[44mwarn\033[0m: "#m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
+#define PRINT_WARN3(m,a,b,c) printf( "\033[44mwarn\033[0m: "#m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
+#define PRINT_WARN4(m,a,b,c,d) printf( "\033[44mwarn\033[0m: "#m" (%s %s)\n", a, b, c, d, __FILE__, __FUNCTION__)
+#define PRINT_WARN5(m,a,b,c,d,e) printf( "\033[44mwarn\033[0m: "#m" (%s %s)\n", a, b, c, d, e, __FILE__, __FUNCTION__)
 
   // Message macros
 #ifdef DEBUG
-#define PRINT_MSG(m) printf( "Stage: "m" (%s %s)\n", __FILE__, __FUNCTION__)
-#define PRINT_MSG1(m,a) printf( "Stage: "m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
-#define PRINT_MSG2(m,a,b) printf( "Stage: "m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
-#define PRINT_MSG3(m,a,b,c) printf( "Stage: "m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
-#define PRINT_MSG4(m,a,b,c,d) printf( "Stage: "m" (%s %s)\n", a, b, c, d, __FILE__, __FUNCTION__)
-#define PRINT_MSG5(m,a,b,c,d,e) printf( "Stage: "m" (%s %s)\n", a, b, c, d, e,__FILE__, __FUNCTION__)
+#define PRINT_MSG(m) printf( "Stage: "#m" (%s %s)\n", __FILE__, __FUNCTION__)
+#define PRINT_MSG1(m,a) printf( "Stage: "#m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
+#define PRINT_MSG2(m,a,b) printf( "Stage: "#m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
+#define PRINT_MSG3(m,a,b,c) printf( "Stage: "#m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
+#define PRINT_MSG4(m,a,b,c,d) printf( "Stage: "#m" (%s %s)\n", a, b, c, d, __FILE__, __FUNCTION__)
+#define PRINT_MSG5(m,a,b,c,d,e) printf( "Stage: "#m" (%s %s)\n", a, b, c, d, e,__FILE__, __FUNCTION__)
 #else
-#define PRINT_MSG(m) printf( "Stage: "m"\n" )
-#define PRINT_MSG1(m,a) printf( "Stage: "m"\n", a)
-#define PRINT_MSG2(m,a,b) printf( "Stage: "m"\n,", a, b )
-#define PRINT_MSG3(m,a,b,c) printf( "Stage: "m"\n", a, b, c )
-#define PRINT_MSG4(m,a,b,c,d) printf( "Stage: "m"\n", a, b, c, d )
-#define PRINT_MSG5(m,a,b,c,d,e) printf( "Stage: "m"\n", a, b, c, d, e )
+#define PRINT_MSG(m) printf( "Stage: "#m"\n" )
+#define PRINT_MSG1(m,a) printf( "Stage: "#m"\n", a)
+#define PRINT_MSG2(m,a,b) printf( "Stage: "#m"\n,", a, b )
+#define PRINT_MSG3(m,a,b,c) printf( "Stage: "#m"\n", a, b, c )
+#define PRINT_MSG4(m,a,b,c,d) printf( "Stage: "#m"\n", a, b, c, d )
+#define PRINT_MSG5(m,a,b,c,d,e) printf( "Stage: "#m"\n", a, b, c, d, e )
 #endif
 
   // DEBUG macros
 #ifdef DEBUG
-#define PRINT_DEBUG(m) printf( "debug: "m" (%s %s)\n", __FILE__, __FUNCTION__)
-#define PRINT_DEBUG1(m,a) printf( "debug: "m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
-#define PRINT_DEBUG2(m,a,b) printf( "debug: "m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
-#define PRINT_DEBUG3(m,a,b,c) printf( "debug: "m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
-#define PRINT_DEBUG4(m,a,b,c,d) printf( "debug: "m" (%s %s)\n", a, b, c ,d, __FILE__, __FUNCTION__)
-#define PRINT_DEBUG5(m,a,b,c,d,e) printf( "debug: "m" (%s %s)\n", a, b, c ,d, e, __FILE__, __FUNCTION__)
+#define PRINT_DEBUG(m) printf( "debug: "#m" (%s %s)\n", __FILE__, __FUNCTION__)
+#define PRINT_DEBUG1(m,a) printf( "debug: "#m" (%s %s)\n", a, __FILE__, __FUNCTION__)    
+#define PRINT_DEBUG2(m,a,b) printf( "debug: "#m" (%s %s)\n", a, b, __FILE__, __FUNCTION__) 
+#define PRINT_DEBUG3(m,a,b,c) printf( "debug: "#m" (%s %s)\n", a, b, c, __FILE__, __FUNCTION__)
+#define PRINT_DEBUG4(m,a,b,c,d) printf( "debug: "#m" (%s %s)\n", a, b, c ,d, __FILE__, __FUNCTION__)
+#define PRINT_DEBUG5(m,a,b,c,d,e) printf( "debug: "#m" (%s %s)\n", a, b, c ,d, e, __FILE__, __FUNCTION__)
 #else
 #define PRINT_DEBUG(m)
 #define PRINT_DEBUG1(m,a)
</code>

majorx234, 2016-08-04:
added the diff file, sorry in the comment it looks realy ugly

[stage.hh.diff.zip](https://github.com/rtv/Stage/files/402145/stage.hh.diff.zip)

kartikmohta, 2016-08-11:
See #62

xuefengchang, 2016-10-25:
which directory should I put the stage.hh.diff file to?

majorx234, 2016-10-25:
you have to apply the patch with the terminal command "patch"! here is an tutorial:
http://www.thegeekstuff.com/2014/12/patch-command-examples/
goodl luck

rtv, 2017-04-05:
applied PR for #62 to fix. Thanks.

closed 2017-04-05 by rtv