rtv/Stage #65: stage command not found

asfandyarkhan01, 2016-09-05:
I followed all the steps of installation but when I want to test by running an example I get a meaasge that "stage command is not found".

rtv, 2016-09-05:
Sounds like stage is not in your path. 
Please follow the instructions for getting help in future. 

Sent from mobile gadget

rish691, 2016-10-18:
I got the same problem,what to do ??

rtv, 2016-10-18:
Add the directory that contains the stage binary to your PATH. This is a
basic UNIX thing, and not Stage-specific.

rish691, 2016-10-18:
Yeah it was working fine,my bad

closed 2017-01-10 by rtv

jandisoon, 2019-03-26:
helloo i have the same problem can you tell me how did you add the stage path ??

haydaradz, 2019-04-08:
hello , tell us how did you add the stage path

Tony-Feng, 2019-05-19:
It's under ~/stage4/libstage/stage

Tony-Feng, 2019-05-19:
If you follow the instruction from the repo, then `export PATH="$PATH:~/stage4/libstage"` should work