rtv/Stage #69: Player/Stage for ubuntu 14 , ubuntu 16 even for windows 10 (in bash)

pinftv, 2016-11-08:
If anyone want's to install it in ubuntu 14 or 16 I have modify the files and successfully installed it in both. I have write an easy instruction . Please contact me for get the files.

Scalpel78, 2016-12-08:
Can you put your changes here???

rtv, 2016-12-08:
Hi.

What do you want me to do?

Richard/

Scalpel78, 2016-12-08:
Hi, this was mostly a comment for user piftv which said he har managed to get this working on Ubuntu 16, but didn't specify how. I'm trying to do the same, but get the following error when trying to run 

```
frode@XenialXerus:~/GIT/stage4$ stage Stage/worlds/simple.world 
stage: error while loading shared libraries: libstage.so.4.1.1: cannot open shared object file: No such file or directory
```

I'm coming here from this thread: http://answers.ros.org/question/249570/gazebo-or-rviz/

I've followed the steps as laid out in INSTALL.txt. After it failed I ran;
`export LD_LIBRARY_PATH=$HOME/GIT/stage4/lib`, and then redid the make-steps. Still same errror.

I've opted to NOT run this full command, since it ends up wanting to uninstall all of ROS Kinetic:
`sudo apt-get install pkg-config libfltk1.1-dev libgl1-mesa-dev libglu1-mesa-dev libpng12-dev libtool`

I do not have Player installed, as I believe it is optional.

Can you see what I'm doing wrong?

jpgr87, 2016-12-08:
How did you install stage to get to that point?  Did you clone the git repository and build it?  Or did you download it from somewhere?

pinftv, 2016-12-08:
Hi guys, i dont know how update the files here I changed some things in source files, but I don't remember what exactly, because I was changing until it compiled.

Scalpel78, 2016-12-08:
pinftv, and that's why it's especially annoying when you add a cryptic message here that you have secret way of doing this. Hopefully next time you'll share what you did instead.

jpgr87, I downloaded from GIT, and followed the instructions in https://github.com/rtv/Stage/blob/master/INSTALL.txt

pinftv, 2016-12-08:
[install player stage.tar.gz](https://github.com/rtv/Stage/files/640452/install.player.stage.tar.gz)

Okey, Igot it. Here are the files, note that some things are only for ubuntu 14 and others for 16 it is in the comments. Good luck. If you have any problem don't hesitate to contact me, if I will know how to help, I will do it.

pinftv, 2016-12-08:
@Scalpel78 I said I didn't know how to upload files. Now I guessed so I shared. I could don't write anything about it, that I got player/stage to work in Ubunu14 and 16 but I wanted to share it. I did the way I knew. :)

amamory, 2016-12-08:
People mix up player, stage, stageros. 
They are 3 different things that happen to 'usually' work together, bit also work standalone.

All the modifications mentioned in this thread are for Player, not for Stage. So **this topic is misplaced**.
I have compiled Stage for 12.04, 14.04 , and 16.04 without a single modification in the source code or cmake scripts.

On the other hand, Player requires lots of modifications.
This link provides patch files for 12.04 and 14.04. In the future i will provide the patch for 16.04.
https://github.com/lsa-pucrs/Player/tree/master/patch/install

Scalpel78, 2016-12-08:
@pinftv Thanks for the file, but this is an older Stage 3.2.2, not Stage 4.1.1 , and that build crashes on my 16.04.

@amamory Thanks for the comment, I'll try to start from scratch, and follow the longer variant of the instructions from the INSTALL.txt file. Maybe I'll find something I missed the first time round.

pinftv, 2016-12-08:
@amamory The file I provided is modificated and it install PLAYER and STAGE just need to follow the instructions. It even works in bash for windows (installation as for ubuntu 14)

pinftv, 2016-12-08:
@Scalpel78 does it throw some error? or just crashes? I have install it in many computers in my University using Ubuntu 16 and it works well. People are using it in their lessons

Scalpel78, 2016-12-08:
@pinftv when running 'make' I get this, and a few other similar errors:
```

/home/frode/Downloads/Stage3/install player stage/Stage-3.2.2-Source/libstage/region.cc:10:1: error: reference to ‘Region’ is ambiguous
 Region::Region( SuperRegion* sr) : 
 ^
In file included from /usr/include/FL/x.H:49:0,
                 from /usr/include/FL/fl_draw.H:27,
                 from /home/frode/Downloads/Stage3/install player stage/Stage-3.2.2-Source/libstage/stage.hh:61,
                 from /home/frode/Downloads/Stage3/install player stage/Stage-3.2.2-Source/libstage/region.hh:8,
                 from /home/frode/Downloads/Stage3/install player stage/Stage-3.2.2-Source/libstage/region.cc:7:
/usr/include/X11/Xutil.h:273:26: note: candidates are: typedef struct _XRegion* Region
 typedef struct _XRegion *Region;
```

pinftv, 2016-12-08:
@Scalpel78 did you install first Player? and everything as is in the instructions?

Scalpel78, 2016-12-08:
@pinftv I tried the full installation guide from your file now. The only step I skipped was to install 'libjpeg62-dev' since it wants to uninstall all of ROS.

When trying to make player it ends up with this:
```
Scanning dependencies of target playerjpeg
[  6%] Building C object libplayerjpeg/CMakeFiles/playerjpeg.dir/playerjpeg.o
[  6%] Linking C shared library libplayerjpeg.so
[  6%] Built target playerjpeg
Scanning dependencies of target playerwkb
[  6%] Building C object libplayerwkb/CMakeFiles/playerwkb.dir/playerwkb.o
[  6%] Linking C shared library libplayerwkb.so
/usr/bin/ld: cannot find -lgeos
collect2: error: ld returned 1 exit status
libplayerwkb/CMakeFiles/playerwkb.dir/build.make:95: recipe for target 'libplayerwkb/libplayerwkb.so.3.0.2' failed
make[2]: *** [libplayerwkb/libplayerwkb.so.3.0.2] Error 1
CMakeFiles/Makefile2:816: recipe for target 'libplayerwkb/CMakeFiles/playerwkb.dir/all' failed
make[1]: *** [libplayerwkb/CMakeFiles/playerwkb.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
frode@XenialXerus:~/src/player-3.0.2/build$ 

```

pinftv, 2016-12-08:
@Scalpel78 try installing  libgeos++-dev ("sudo apt-get install  libgeos++-dev")

Scalpel78, 2016-12-08:
@pinftv That fixed that error. Got further now. Got a lot of warning because of old Boost version, but seems to be compiled.

I skipped installing 'libfltk1.1 libfltk1.1-dev' since they want to remove ROS.

When running `grep -rl 'fPIE' ./ | xargs sed -i 's/fPIE/fPIC/g'` I got `sed: no input file`. Not sure if that is relevant.

But, unfortunately I still get this when making Stage:
```
[ 31%] Building CXX object libstage/CMakeFiles/stage.dir/option.o
[ 32%] Building CXX object libstage/CMakeFiles/stage.dir/powerpack.o
[ 34%] Building CXX object libstage/CMakeFiles/stage.dir/region.o
/home/frode/src/Stage-3.2.2-Source/libstage/region.cc:10:1: error: reference to ‘Region’ is ambiguous
 Region::Region( SuperRegion* sr) : 
 ^
In file included from /usr/include/FL/x.H:49:0,
                 from /usr/include/FL/fl_draw.H:27,
                 from /home/frode/src/Stage-3.2.2-Source/libstage/stage.hh:61,
                 from /home/frode/src/Stage-3.2.2-Source/libstage/region.hh:8,
                 from /home/frode/src/Stage-3.2.2-Source/libstage/region.cc:7:
/usr/include/X11/Xutil.h:273:26: note: candidates are: typedef struct _XRegion* Region
 typedef struct _XRegion *Region;
                          ^
In file included from /home/frode/src/Stage-3.2.2-Source/libstage/region.cc:7:0:
/home/frode/src/Stage-3.2.2-Source/libstage/region.hh:61:9: note:                 class Stg::Region
   class Region
         ^
/home/frode/src/Stage-3.2.2-Source/libstage/region.cc:17:1: error: reference to ‘Region’ is ambiguous
 Region::~Region()
 ^
In file included from /usr/include/FL/x.H:49:0,
                 from /usr/include/FL/fl_draw.H:27,
                 from /home/frode/src/Stage-3.2.2-Source/libstage/stage.hh:61,
                 from /home/frode/src/Stage-3.2.2-Source/libstage/region.hh:8,
                 from /home/frode/src/Stage-3.2.2-Source/libstage/region.cc:7:
/usr/include/X11/Xutil.h:273:26: note: candidates are: typedef struct _XRegion* Region
 typedef struct _XRegion *Region;
                          ^
In file included from /home/frode/src/Stage-3.2.2-Source/libstage/region.cc:7:0:
/home/frode/src/Stage-3.2.2-Source/libstage/region.hh:61:9: note:                 class Stg::Region
   class Region
         ^
libstage/CMakeFiles/stage.dir/build.make:638: recipe for target 'libstage/CMakeFiles/stage.dir/region.o' failed
make[2]: *** [libstage/CMakeFiles/stage.dir/region.o] Error 1
CMakeFiles/Makefile2:124: recipe for target 'libstage/CMakeFiles/stage.dir/all' failed
make[1]: *** [libstage/CMakeFiles/stage.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

pinftv, 2016-12-08:
with Stage you have a conflict because X11 use a library with the same name

Scalpel78, 2016-12-08:
Yes, it's running in the build directory, but I'm still getting same error.

```
frode@XenialXerus:~/src/Stage-3.2.2-Source/build$ ls ./
assets          cmake_install.cmake      CTestTestfile.cmake  libstageplugin  worlds
CMakeCache.txt  CPackConfig.cmake        examples             Makefile
CMakeFiles      CPackSourceConfig.cmake  libstage             stage.pc
frode@XenialXerus:~/src/Stage-3.2.2-Source/build$ grep -rl 'fPIE' ./ | xargs sed -i 's/fPIE/fPIC/g'
sed: no input files
frode@XenialXerus:~/src/Stage-3.2.2-Source/build$ 
```
What is the purpose of that command? I don't quite understand it.

pinftv, 2016-12-08:
@Scalpel78 I have checked and it is not necessary you can skip it I see it remains changed. the purpose was to change compiling/linking options.

Scalpel78, 2016-12-08:
Ok, thanks for your help. Then I'm giving up trying to get this fileset working, and instead focus on trying to get the current Stage version from GitHub working.

pinftv, 2016-12-08:
@Scalpel78 but player works?

Scalpel78, 2016-12-08:
@pinftv I think so, but not sure what the easiest way to test it is?

If I just run player I get this, so I assume it's working?
```
frode@XenialXerus:~/src/player-3.0.2$ player
Syntax: python player.py imagefile(s)
```

pinftv, 2016-12-08:
@Scalpel78 you can do "sudo apt-get purge x11" then compile stage

Scalpel78, 2016-12-08:
No thanks, I think I'm giving up on this. This was just too complicated getting to work. I think I'll go back to researching how I can use Gazebo/Rviz instead. Seems like a lot more stable solution than Player/Stage, which it seems hasn't really been updated for many years.

jpgr87, 2016-12-08:
For what it's worth, if you're installing ros from debs provided at packages.ros.org, you can install a stage deb from there:

http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-stage/

These packages put stage at /opt/ros/kinetic/bin/stage

Scalpel78, 2016-12-09:
The irony here is that I had that package installed all along, just didn't know about it... It's probably part of the 'ros-kinetic-desktop-full' package. I can now start Stage 4.1.1!

closed 2017-01-10 by rtv