rtv/Stage #83: stage Stage/worlds/simple.world stage: error while loading shared libraries:

KeyvanDerakhshanNik, 2017-05-24:
Hi 
I am pretty new in the stage, I used fast installation in http://rtv.github.io/Stage/install.html
My OS is Ubuntu. When I Started Stage "stage Stage/worlds/simple.world". 
I took this error : 
libstage.so.4.3.0: cannot open shared object file: No such file or directory
could you please  help me to recover this problem. 
BR

rtv, 2017-05-24:
Did you do step 4 (note the instructions look a little garbled in HTML due a messed up _ character - look at the source in an editor). The  problem is probably that your LD_LIBRARY_PATH does not contain the directory containing libstage.so.4.3.0.  Or maybe you didn't make install.

KeyvanDerakhshanNik, 2017-05-24:
Thanks, I reinstall it by sudo command, It is worked now.

tmozgach, 2018-01-19:
Another solution:
```
cd /
sudo find ./ | grep libstage.so
./home/tatyana/stg/lib/libstage.so
./home/tatyana/stg/lib/libstage.so.4.3.0
./home/tatyana/stage4/libstage/libstage.so
./home/tatyana/stage4/libstage/libstage.so.4.3.0

LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/tatyana/stg/lib/:/home/tatyana/stage4/libstage/
export LD_LIBRARY_PATH
$HOME/stg/bin/stage $HOME/stage4/Stage/worlds/simple.world

```

Herrsun, 2021-01-12:
I also got the same Error, but even though the LD_LIBRARY_PATH is right, it does't work.
But I found, wenn I installed the libpng16-dev instead of libpng12-dev package by `pip install libpng16-dev`, the porblem was solved. `stage Stage/worlds/simple.world` can be run flowlessly.