rtv/Stage #89: Unable to set 'speedup' running headless (ie. --gui)

jacobperron, 2018-04-26:
We can set the desired real-time factor (aka. speedup) by placing the following in our world file:

        # Set 10x real-time
        speedup 10 

But this seems to only work when running with the GUI enabled. As soon as I run headless (e.g. `stage --gui my.world`), the simulation seems to run as fast as possible (as if `speedup 0` was set).

It would be nice to be able to set the speedup factor regardless of whether the GUI is enabled. This would allow running headless simulations with distributed systems (e.g. ROS) with processes that may not be able to keep up with how fast the simulation is progressing.

rtv, 2018-04-26:
That’s an interesting use case I’ve never considered. Will think about it.

tylertian123, 2024-09-30:
This would definitely be useful. I had an issue where the GUI was extremely slow on larger maps, but I was also using it with ROS so running headless at full speed would be unideal. I'm not sure if this project is still being maintained but I made a PR that adds support for `speedup` in headless. Hope someone finds this useful!