rtv/Stage #90: Player make error in Ubuntu 16.04

Aaricis, 2018-05-03:
I make Player in Ubuntu 16.04, but it can not success.
[image]
what should I do?

Aaricis, 2018-05-03:
Information like this
`Scanning dependencies of target playerc++
[ 14%] Building CXX object client_libs/libplayerc++/CMakeFiles/playerc++.dir/playerc++.o
In file included from /home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerclient.h:66:0,
                 from /home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerc++.h:61,
                 from /home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerc++.cc:44:
/usr/include/boost/signal.hpp:17:4: warning: #warning "Boost.Signals is no longer being maintained and is now deprecated. Please switch to Boost.Signals2. To disable this warning message, define BOOST_SIGNALS_NO_DEPRECATION_WARNING." [-Wcpp]
 #  warning                  "Boost.Signals is no longer being maintained and is
    ^
[ 14%] Building CXX object client_libs/libplayerc++/CMakeFiles/playerc++.dir/playerclient.o
In file included from /home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerclient.h:66:0,
                 from /home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerc++.h:61,
                 from /home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerclient.cc:54:
/usr/include/boost/signal.hpp:17:4: warning: #warning "Boost.Signals is no longer being maintained and is now deprecated. Please switch to Boost.Signals2. To disable this warning message, define BOOST_SIGNALS_NO_DEPRECATION_WARNING." [-Wcpp]
 #  warning                  "Boost.Signals is no longer being maintained and is
    ^
In file included from /usr/include/pthread.h:24:0,
                 from /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:35,
                 from /usr/include/x86_64-linux-gnu/c++/5/bits/gthr.h:148,
                 from /usr/include/c++/5/ext/atomicity.h:35,
                 from /usr/include/c++/5/bits/ios_base.h:39,
                 from /usr/include/c++/5/ios:42,
                 from /usr/include/c++/5/ostream:38,
                 from /usr/include/c++/5/iostream:39,
                 from /home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerclient.cc:48:
/home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerclient.cc: In member function ‘void PlayerCc::PlayerClient::RunThread()’:
/home/zhangkaige/player-3.0.2/client_libs/libplayerc++/playerclient.cc:171:34: error: expected unqualified-id before numeric constant
     boost::xtime_get(&xt, boost::TIME_UTC);
                                  ^
client_libs/libplayerc++/CMakeFiles/playerc++.dir/build.make:86: recipe for target 'client_libs/libplayerc++/CMakeFiles/playerc++.dir/playerclient.o' failed
make[2]: *** [client_libs/libplayerc++/CMakeFiles/playerc++.dir/playerclient.o] Error 1
CMakeFiles/Makefile2:531: recipe for target 'client_libs/libplayerc++/CMakeFiles/playerc++.dir/all' failed
make[1]: *** [client_libs/libplayerc++/CMakeFiles/playerc++.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
`

Ederwardlu, 2020-03-31:
i have same problems,when modify ''boost::xtime_get(&xt, boost::TIME_UTC);'' as''boost::xtime_get(&xt, boost::TIME_UTC_);'' it can go to 78%