rtv/Stage #91: How can I install "Player" in Ubuntu 16.04?

Aaricis, 2018-05-03:
I met some errors.
[image]
It told me `/usr/bin/ld:  cannot find -lgeos`, but I have installed `libgeos.so` already.
[image]
How should I do?

rtv, 2018-05-03:
Stage doesn’t use libgeos. You are building Player.

rtv, 2018-05-04:
> Thanks for your reply.
> when I making "Player", it told me /usr/bin/ld: cannot find -lgeos. but I
> have installed libgeos.so already.
> Why dose it cannot be find? How do I resolve this error?

https://stackoverflow.com/questions/16710047/usr-bin-ld-cannot-find-lnameofthelibrary

Aaricis, 2018-05-04:
Thanks a lot.
error `/usr/bin/ld: cannot find -lgeos` has been resolved.
But there is an another error `/home/zhangkaige/player/server/drivers/camera/kinect/kinectdriver.cc:116:37: fatal error: libfreenect/libfreenect.h: No such file or directory
`
[image]

I have checked `kinectdriver.cc`. It shows me this:
[image]
I try to find some solutions but they do not work.

Aaricis, 2018-05-04:
[image]
[image]

rtv, 2018-05-04:
I don't know how to fix the Player Kinect driver problem. It is not related
to Stage. Please seek help with Player on the player forum. And don't post
images of text on ANY forum!

closed 2019-02-10 by rtv