rtv/Stage #99: install player/stage on ubuntu18.04

bengao1, 2020-03-21:
Hi!

I try to install player/stage on ubuntu  v18.04.

http://playerstage.sourceforge.net/

But I can't because of its missing some package and I can't resolve it!

For example, impossible to install robot-player

Anybody did it?

I need this application.

Thank's a lot

amamory, 2020-03-22:
Player and Stage are highly configurable. Thus, the dependency list will
'depend' on the features you choose to compile.
ccmake will help you to select these features. I am assuming you are
familiar with make/cmake ....

I have a project using player and stage here
https://github.com/lsa-pucrs/donnie-assistive-robot-sw/blob/devel/3rd-party/player-stage.sh

If you see, there is a long list of features that we can enable/disable.
Other than that, as the OS evolve, the libs get renamed/updated.
So it is just a matter of finding what is the current lib name  for the OS
you are using.
Something the libs change their internal headers, functions, etc. Then you
have to go to Player/Stage source code to update it, one by one.
However, this case is quite rare.

In summary, it is not that difficult to update Player/Stage to another OS
version. Just, Ask google and it will tell you.

ps.: by far, the most problematic Player/Stage modules are those related to
OpenCV. If you dont have much of experience with OpenCV, I would recommend
just to disable these modules.

I hope it will help you!
Alexandre Amory

bengao1, 2020-03-23:
Same problem

`mkdir -p build
cd build
cmake .. `

result :
-- Player version 3.0.2
-- Operating system is Linux
-- Got System Processor x86_64
-- Linux x86_64 Detected
-- Player will be installed to /usr/local
-- Player libraries will be in lib64
-- Large file support is disabled.
-- ===== Client Libraries =====
-- Python bindings for C client library will be built
Python 3.7.4
CMake Error at client_libs/libplayerc/bindings/python/CMakeLists.txt:56 (STRING):
  STRING sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.

-- Ruby bindings for C client library will be built
-- Ruby bindings for C client will be installed to lib64/x86_64-linux-gnu/site_ruby
-- Python bindings for C++ client library will not be built - disabled by user
-- Ruby bindings for C++ client library will be built
-- Boost version: 1.65.1
-- PlayerC++ client library will be built with Boost::Thread support.
-- PlayerC++ client library will be built with Boost::Signals support.
-- ============================
-- 
-- ----- Processing driver make files -----
-- Checking for module 'artoolkitplus>=2.1'
--   No package 'artoolkitplus' found
-- Checking for module 'gstreamer-base-0.10'
--   No package 'gstreamer-base-0.10' found
-- Checking for module 'GbxGarminAcfr=1.0.0'
--   No package 'GbxGarminAcfr' found
-- Checking for module 'libstatgrab'
--   No package 'libstatgrab' found
-- Checking for module 'libgarcia'
--   No package 'libgarcia' found
-- Checking for module 'flexiport=1.0.0'
--   No package 'flexiport' found
-- Checking for module 'GbxSmartBatteryAcfr=1.0.0'
--   No package 'GbxSmartBatteryAcfr' found
-- Checking for module 'GbxSickAcfr=1.0.0'
--   No package 'GbxSickAcfr' found
-- Checking for module 'hokuyo_aist=1.0.0'
--   No package 'hokuyo_aist' found
-- Checking for module 'howl>=0.9.6'
--   No package 'howl' found
-- ----- Done -----
-- 
-- ===== Drivers =====
-- The following drivers will be built:
-- AioToSonar
-- accel_calib
-- acts
-- alsa
-- amcl
-- amtecpowercube
-- aodv
-- bitlogic
-- blobposition
-- blobtodio
-- blobtracker
-- bumper2laser
-- bumpersafe
-- bumpertodio
-- camera1394
-- cameracompress
-- camerauncompress
-- camerauvc
-- camerav4l2
-- camfilter
-- canonvcc4
-- clodbuster
-- cmdsplitter
-- cmucam2
-- cmvision
-- create
-- cvcam
-- deadstop
-- diocmd
-- diodelay
-- diolatch
-- dummy
-- epuck
-- er1
-- erratic
-- fakelocalize
-- festival
-- flockofbirds
-- garminnmea
-- globalize
-- goto
-- gridmap
-- gripcmd
-- imageseq
-- imgcmp
-- imgsave
-- inhibitor
-- insideM300
-- iwspy
-- kartowriter
-- khepera
-- laserbar
-- laserbarcode
-- lasercspace
-- lasercutter
-- laserposeinterpolator
-- laserptzcloud
-- laserrescan
-- lasersafe
-- lasertoranger
-- linuxjoystick
-- linuxwifi
-- localbb
-- mapcspace
-- mapfile
-- mapscale
-- mbicp
-- mica2
-- microstrain
-- motionmind
-- mricp
-- nd
-- nomad
-- obot
-- opaquecmd
-- p2os
-- passthrough
-- pbslaser
-- portio
-- postgis
-- postlog
-- ptu46
-- rangerposeinterpolator
-- rangertodio
-- rangertolaser
-- readlog
-- relay
-- rflex
-- roboteq
-- robotracker
-- roomba
-- rs4leuze
-- rt3xxx
-- searchpattern
-- segwayrmp400
-- serialstream
-- serio
-- shapetracker
-- sickLDMRS
-- sickRFI341
-- sicklms200
-- sicklms400
-- sicknav200
-- sicks3000
-- simpleshape
-- skyetekM1
-- snd
-- sonartoranger
-- sonyevid30
-- speechcmd
-- sphereptz
-- stalltodio
-- suppressor
-- tcpstream
-- upcbarcode
-- vec2map
-- velcmd
-- vfh
-- videocanny
-- vmapfile
-- wavefront
-- wbr914
-- writelog
-- 
-- The following drivers will not be built:
-- acr120u - Could not find header usb.h
-- amtecm5 - Could not find header Device.h
-- artoolkitplus - Could not find package artoolkitplus
-- cameragst - Could not find package gstreamer-base-0.10
-- camerav4l - Could not find header linux/videodev.h
-- eedhcontroller - Disabled - probably doesn't build
-- flexiport - Could not find package flexiport
-- garcia - Could not find package libgarcia
-- gbxgarminacfr - Could not find package GbxGarminAcfr
-- gbxsickacfr - Could not find package GbxSickAcfr
-- hokuyo_aist - Could not find package hokuyo_aist
-- isense - Could not find header isense/isense.h
-- laservisualbarcode - Has not been updated to use dynamic message structures
-- laservisualbw - Has not been updated to use dynamic message structures
-- nimu - Could not find header usb.h
-- oceanserver - Could not find package GbxSmartBatteryAcfr
-- phidgetAcc - Could not find header phidget21.h
-- phidgetRFID - Could not find header phidget21.h
-- phidgetifk - Could not find header phidget21.h
-- rcore_xbridge - Could not find header libparticle.h
-- reb - Disabled by default
-- robotino - Could not find header robotinocom.h
-- segwayrmp - Disabled by default
-- service_adv_mdns - Disabled by default
-- sphere - Could not find header linux/videodev.h
-- sphinx2 - Disabled by default
-- sr3000 - Could not find header libusbSR.h
-- statgrab - Could not find package libstatgrab
-- stoc - Could not find header SVS/svsclass.h
-- swissranger - Could not find header libMesaSR.h
-- unicapimage - Could not find header unicap.h
-- xsensmt - Disabled by default - set XSENSMT_SOURCE_DIR
-- yarpimage - Could not find header yarp/os/all.h
-- ===================
-- 
-- ===== Examples =====
-- ====================
-- 
-- ===== Player utilities =====
-- XMMS plugin will not be built - XMMS not found
-- ============================
-- 
-- Configuring incomplete, errors occurred!
See also "/home/use/Téléchargements/player-3.0.2(1)/player-3.0.2/build/CMakeFiles/CMakeOutput.log".
See also "/home/use/Téléchargements/player-3.0.2(1)/player-3.0.2/build/CMakeFiles/CMakeError.log".

amamory, 2020-03-23:
use ccmake to disable python and ruby bindings if you dont plan to use them.
search google about how to install ccmake and explore it.

have fun !

bengao1, 2020-03-24:
Tkanks but i don't find the issue

bengao1, 2020-03-24:
Tkanks a lot
Now 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/use/Téléchargements/player-3.0.2(1)/player-3.0.2/build

but after sudo make install :

> [  1%] Generating functiontable_gen.h

> /bin/sh: 1: Syntax error: "(" unexpected

> libplayerinterface/CMakeFiles/functiontable_gen.dir/build.make:101: recipe for target 'libplayerinterface/functiontable_gen.h' failed

> make[2]: *** [libplayerinterface/functiontable_gen.h] Error 2

> CMakeFiles/Makefile2:180: recipe for target 'libplayerinterface/CMakeFiles/functiontable_gen.dir/all' failed

> make[1]: *** [libplayerinterface/CMakeFiles/functiontable_gen.dir/all] Error 2

> Makefile:151: recipe for target 'all' failed

> make: *** [all] Error

richmattes, 2020-03-24:
It looks like you're dealing with a Player issue, not a Stage issue.  Please report Player issues to the Player project
https://github.com/playerproject/player

bengao1, 2020-03-24:
sorry but what do i have to do?

bengao1, 2020-03-24:
I need some patch?

bengao1, 2020-03-24:
thanks al lot.
but after sudo make install

> CMake Error at server/cmake_install.cmake:50 (file):

  

> file RPATH_CHANGE given FILE "/usr/local/bin/player" that does not exist.

> Call Stack (most recent call first):

 

>  cmake_install.cmake:57 (include)

> Makefile:85: recipe for target 'install' failed

> make: *** [install] Error 1

bengao1, 2020-03-25:
I had a old /usr/local/bin/player that I have delated 
After reinstalling player, I think now it's OK.

Thank's to everybody's

bengao1, 2020-03-29:
to continue with program

In my old version, I defined a sonar, but i understand that sonar is now obsolete, so I take ranger :

In my world.cfg :

driver
(		
  name "stage"
  provides [ "6665:position2d:0" "6665:ranger:0" "6665:blobfinder:0" ]
  model "Penelope" 
)

In my main.cc :

       using namespace PlayerCc;
     
        PlayerClient robot(gHostname, gPort);
        
        Position2dProxy pp(&robot, 0);  
        
        RangerProxy sonarProxy(&robot,0);   

No problem with compilation , I can't  run player world.cfg

But when I run ./main -p6665

playerc warning   : warning : [Player v.3.1.1-dev] connected on [localhost:6665] with sock 3

playerc error   : socket disconnected
playerc warning   : warning : failed to reconnect
playerc warning   : warning : no socket to write to
playerc warning   : warning : no socket to write to
playerc error   : close failed with error [Bad file descriptor]
PlayerClient::Read()(-1) : warning : failed to reconnect

Any Idea?

closed 2020-03-29 by bengao1